tinue/apa102-pi #3: Change GPIO Output

must4rd, 2016-01-30:
Hey i am new to this. 
is it possible to change the GPIO output? 
Because in addition to dotstars i have hyperion running and hyperion is using these pins.
Adafruits Raspberry Library is doing it like this:
`
import time
from dotstar import Adafruit_DotStar

numpixels = 64 # Number of LEDs in strip

# Here's how to control the strip from any two GPIO pins:
datapin   = 23
clockpin  = 24
strip     = Adafruit_DotStar(numpixels, datapin, clockpin)
`

Could you explain how to change it in your code?

tinue, 2016-01-30:
These GPIO pins form the only hardware SPI port on the Raspberry Pi. Which is why Hyperion uses the same pins.

Theoretically one can handle the SPI protocol in software using any two GPIO pins (usually called "bit banging"). This is what the Adafruit library does. With my particular strip I had little success with the Adafruit library if I didn't use the hardware GPIO: There were many disturbances visible.

The SPI library that I use (spidev) does not support this, though. It can only use the hardware SPI port. Therefore I am sorry to disappoint you: You can't change the GPIO ports that are used.

closed 2016-01-30 by tinue

must4rd, 2016-01-30:
ok nice to know...
so i will have to use the adafruit library and have to write my own script :-/
Damn! because you did a nice job here and your effects are looking awesome

tinue, 2016-01-30:
Yes... You might still be able to use colorschemes.py and colorcycletemplate.py: You would have to rename a few method calls, but they are not fundamentally different.