tinue/apa102-pi #9: SPI closed?

estiens, 2016-06-12:
I have tried this wired up to a logic converter and directly interfacing the MOSI and CLOCK pins to the LEDs on a Raspberry Pi 3 -- If I try the Adafruit libraries it runs, but nothing happens. If I try running your program I get the following error message and then it shuts down.

I have enabled SPI through raspi-config and by manually loading with `sudo modprobe spi_bcm2835` (which runs without errors) -- I tested the SPI with a C program and it is sending and receiving signals...

Any ideas?

```Shutdown not implemented
Strip cleared
SPI closed```

JeroenButer, 2016-06-28:
I encountered the same problem. I have no clue why it thinks SPI is closed, while if tested, SPI works fine. Any help would be much appreciated!

estiens, 2016-06-28:
For me this ended up being an issue with the actual LED strip I was using. When swapped out the library worked fine.

tinue, 2016-07-03:
Sorry, I keep missing the notifications from Github that a new issue has been opened!

What you see are not error messages. These are rather a sort of debug messages that are always shown. If e.g. "runColorCycle.py" the five different color programs are executed, and one should see 5 times the message "SPI closed". This message is printed at the end of the color program when the SPI bus is getting closed.

If the strip remains dark, then it is most likely a hardware issue: Wrong wiring, missing voltage converter, wrong voltage to the strip, broken strip etc.

Also please keep in mind that I have written the library mainly for educational purposes. Initially it was for myself: I wanted to understand how the APA102 protocol works. Then I thought it might be useful for the people as well. If all you need is a working, tested, no-issue library then I would suggest to go with the one from Adafruit.

I am closing there issue because I don't think there is anything that I could in the library itself: It seems to be hardware (one is confirmed to be hardware by @estiens, and I suspect the other one is a hardware issue as well).

closed 2016-07-03 by tinue