tinue/apa102-pi #10: All LEDs just turn white

agarsearcher1985, 2016-07-27:
Hey,

tested your library to get my dotstar leds working but they're just turning all white as soon as I start you program or any other way of setting the color (e.g. strandtest.py). They are RGB LEDs and working because if I connect them to my pi sometimes they are randomly red, green, blue or white.
Do you guys have any idea why they're turning white? I'm using a 3.3V to 5V converter.
PS: I'm using a RPi2 and 240 LEDs

Thanks for your help

jmb, 2016-09-25:
The first part of the demo does turn all the LEDs white for 1 second.

I connected my string of LEDs up as per the Wiring.jpg, enabled SPI, changed the numLEDs in runColorCycle.py and then ran it and it worked first time. I have another simple demo which fades the whole strip through the colour spectrum: https://gist.github.com/jmb/02b8a0a1c7ae5e2b6d8d0103b33159ee

tinue, 2016-09-25:
I did answer previously in German with a private mail. I did not hear back, so I don't know if any of my hints worked. 
Anyway: It seems to work for other people (thanks jmb), so I am closing the issue.

closed 2016-09-25 by tinue

ryan11lb426, 2016-11-07:
I am having the same issue currently