tinue/apa102-pi #11: Power the Pi from same power supply as LEDs, but through USB Port?

stahlfabrik, 2016-09-03:
Hi,

I am currently ordering stuff to recreate your build, in form of a 16x16 LED matrix. I want to power the LEDs and the Pi from the same power supply. I also will use the same logic level power converter as you do. Could you help me? I wonder if it is hazardous for the Pi to be powered via a cut-off micro-USB cable that is attached to the same power supply as the LEDs. Of course, VCC on the level converter would still get its 5 volts from the supply - NOT from the 5V Pi-pin.

I have yet to find an instruction that does this setup. All I can find either power the Pi over its 5v pin (which is said to be unsafe because there are not protective circuits), or by a separate power supply.

I want both: safety by the USB circuits AND only one shared power supply. Yet I do not want to kill the Pi.

Best regards, you have done great work there!

tinue, 2016-09-03:
Thanks for the compliment! At least on the Pi Zero, there seems to be no protection on the USB power port. So it doesn't matter whether you use the 5V pin or the USB input: They are electrically directly connected. 
If you use a strong power supply you could add a capacitor between the 5V pin and ground. This would dampen an initial power surge that the power supply might produce.
FYI: My first Pi Zero fried, and I still have no idea what happened. Maybe something is wrong with my wiring, but maybe the Pi was just from a bad batch. When the days get shorter again (now I don't really need the light) I will probably solder a new one to the LEDs and try again.

stahlfabrik, 2016-09-03:
Oh interesting that it fried your pi. 

I just received a Pi 3 because I did want to have wifi already built in. So I do hope to use the USB port to power it. 

Btw I am planning to put a 1000uF cap between led 5v and ground. Just as adafruit suggests. 

Btw: adafruit recommends putting 330 or 470 ohm to the dataline of neopixels. When they talk about that voltage level shifter they do not talk and use resistors for data in and clock. Do we not need resistors when using the TI chip?

So many questions:-)

tinue, 2016-09-03:
The resistors are for Neopixels, I don't think they are needed for APA102 LEDs (or Dotstars, that's how Adafruit calls them). Plus I don't think a resistor would be required after a level shifter anyway.

The only sure thing that kills a Pi is pulling the 5V for the LEDs from the 5V Pin on the Pi: They draw way too much power and will fry the regulator. 

But if you connect the power supply both to the LEDs and to the Pi it should be fine. Personally I would not bother with the USB connector and power the Pi directly to the 5V pin. I can't see how there can be any sort of protection on the USB input. If the 5V pin has the same voltage as the USB power supply, then there can't be a regulator or a diode in place: Both would reduce the voltage to a lower value. But without schematics it's hard to tell.

Good luck with the Pi 3 WLAN: So far I was unable to keep the WLAN up for more than a few hours. After a while it always locks up. Maybe they fixed this in the meantime, I haven't checked for a while.

stahlfabrik, 2016-09-10:
FYI: My Pi 3 ran flawlessly for 5 days now. Idle all the time. Wifi was always ready. 
I'm still waiting for my leds though:-)
Regarding safety of methods to power the pi: there are big difference it seems. 
An example source: https://www.modmypi.com/blog/how-do-i-power-my-raspberry-pi

tinue, 2016-09-10:
Good to hear that Wifi runs fine on your unit! Maybe they indeed fixed this issue.

About the power: "the Arduino can accept a range of voltage inputs (6 – 20V), and regulates this to desired level internally on board". That's just my point: If you need 5V *regulated*, then you have to supply at least 1V more, i.e. 6V for the Arduino (6V ist the bare minimum, Arduino recommends 7V for consistent results).  If the Arduino is powered via USB instead, then the *source* must be regulated according to the Arduino web site: There is no regulator or protection on the USB power port.

So I still suspect that they are plain wrong when they say about the Raspberry Pi "It should be noted that, unlike the Micro-USB port, there is no regulation or fuse protection on the GPIO". I still believe there is no regulation on USB either.

What can be said is this: A USB power supply "knows" it has to deliver regulated power. It may therefore have surge protection built in. A non-USB 5V power supply might be less safe to use, and might for example produce a power surge when it is switched on. Soldering a USB plug to the supposedly less-safe non-USB power supply will therefore not solve the problem. If you use such a power supply you might just as well connect it to the GPIO pin.

dersch81, 2016-09-19:
Hi, I also ordered all the parts to use this great project. I need 9 meters of APA102 and ordered with 60leds/m. Therefore i have ordered a 5v 30A power supply. I'm planning to power on the pi zero with the same power source. Do you suggest to use the 1000uF capacitor for the pi zero? And maybe is it possible for you to create a wiring plan as a draft and not as a picture? I understand how to wire the 74AHCT125 but where i need to connect the power supply? That i really don't understand without the parts. 

cheers

tinue, 2016-09-25:
Hi, I am really not an electrical engineer, so I am the wrong guy to ask ;) My first Pi Zero stopped working for unknown reasons. I shut it down, unplugged, and a few days later when I wanted to boot it up again it simply did not work anymore. Instead, it quickly became extremely hot. I don't know if it was simply broken, or if something with my wiring is not good. To the best of my knowledge, a capacitor does not hurt at all. So I would actually add one; I certainly will add one if/when I try to reboot this project. Through the summer I had other things to do...

closed 2016-09-25 by tinue