tinue/apa102-pi #22: Should I make a pull request for features I have added?

AlexMeadows, 2017-05-20:
I have been using the APA102_Pi library for a project and have added some new capabilities to it, and I have more features that I plan adding as I have motivation.

I am not sure which of my changes would be beneficial to create pull requests for. [Here is a link to the branch with the changes.](https://github.com/AlexMeadows/APA102_Pi/tree/lightServer) Let me know what you think about it and what I should make pull requests for.

I have a web interface that I use to send JSON messages to the Pis with the strips. I am working on cleaning it up and will probably post to GitHub if there is interest.

tinue, 2017-05-20:
Thanks for your interest! I had a quick look at your changes, but don't fully understand all of them yet ;) 
I will have to actually download and run your branch on my strips to see what it does. This might take a while, because I can't work on this project every weekend.
I have originally written the library to learn python, and to understand how the APA102 LEDs work. So I see it more of an educational tool than a real library to use for projects. So I am mostly interested in adding "educational" stuff, and of course also fun stuff. On first sight, the threading part is very interesting, and the "strobe" part might be fun (that's the part I need to see first in order to understand it). The client/server part is of course interesting and useful, but it might be better of on it's own than being part of the LED library. 
Anyway, these are my initial thoughts. I will definitely check out all of your changes and comment some more here. Thanks again and best regards, Martin

AlexMeadows, 2017-05-20:
Yeah, there are a lot of my changes that are specific to my usage case which is why I tried to keep that code as separate as possible.

The library is nicely made and I am still trying to understand exactly how the APA102 LEDs work. I might work on adding color palettes for patterns and a simulator for displaying led strip colors in the console so I can test changes without a Pi, strip, and all that stuff. However, like you I have other things to do an don't feel like working on it very often.

closed 2017-11-08 by AlexMeadows