tinue/apa102-pi #23: [Q] SK9822 support

Benik3, 2017-07-17:
Hello.

Does the library supports also SK9822 LED strips?
They are better then APA102, but has a slightly different communication. 

Also does anyone tested strip with 600LEDs? We plan to use 20m * 30LED strip controlled by Raspberry.

Thank you :)

tinue, 2017-07-17:
Hi, I don't have any of these, so I guess they would not work. According to this [blog](https://cpldcpu.com/2016/12/13/sk9822-a-clone-of-the-apa102/) it's not too complicated to modify the driver. All that's required is an additional bunch of zero bytes at the end. Why don't you give it a try, and if it works create a pull request?

As for the number of LEDs: The driver itself has a limit of 1024 LEDs at the moment, which could easily be overcome with a few additional lines of code in the "show" method. 

Whether the result looks good or not depends on what you want to do with the LEDs. The APA102 keep their color once they got their data bytes. So the strip will never flicker due to update delays. What does take time is updating the strip. There is a lag between the first and the last LED on the strip to get a new value. You would see this most if you update all of the LEDs to a new color simultaneously.

Benik3, 2017-07-17:
Thanks for the reply.
Well I'm starting with python, but I will see :)
Also I found, that it's still pretty hard to buy them (I can't get them from aliexpress, ebay and similar shops).
The strip will make simple pattern like one colour "snake" etc. so little delay will not be problem.
Also SK9822 can be driven up to 30MHz, but I don't know what is the limit of Rpi.
I'm pretty new in this, so I'm sorry for my stupid question :)

tinue, 2017-07-17:
Your comment about the maximum speed of the chip made me realize a mistake in my last answer: What is slow (due to the driver being written in Python) is updating the buffer for the next "frame". Because of the APA102 keeping their color, this time lag is completely invisible.

Copying the buffer to the LED Strip is then quick: The clock speed used by the driver is 8MHz, but according to [this](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md) it could be set to a higher value. But even with 8MHz and 600 LEDs, this only takes 2-3 Milliseconds.

Anyway. I think the only change required is four more zero bytes at the end, i.e. in `clock_end_frame ` add four to the number of bytes:
```
        for _ in range(4+(self.num_led + 15) // 16):
            self.spi.xfer2([0x00])
```

Because I can't test this I don't want to put it in the code just like that. But if you have the strips, and if you can confirm that it works, I can update the code.

closed 2018-03-09 by tinue

arbolis, 2019-10-19:
I've tried your code @tinue but I don't get a different behavior. I have a 30 LED strip, and only the 1st LED turns red when I run runcolorcycle.py, which is definitely not the expected behavior.
At some point I was able to get all LEDs turned on, most of them being white, but some of them at seemingly random colors. I took some pictures of my setup at https://imgur.com/a/c3elOX1.

tinue, 2019-10-19:
You seem to be using GPIO Pins 20 and 21. Is there a particular reason that you don't use the hardware SPI? At the very least it would be faster compared to bit-banging via regular GPIO pins.

arbolis, 2019-10-19:
@tinue I am a complete noob, I've bought the rpi and the LED strip only for that purpose (to control each LED individually). Are you suggesting I should plug the cables into the pins 19 and 23 in that diagram?: https://pinout.xyz/pinout/

tinue, 2019-10-19:
There is a diagram in my readme file; I suggest you go through the entire readme and see what needs to be done.