tinue/apa102-pi #24: Trying to modify the code as a northern lights show.

raspberrypilights, 2017-09-08:
Hello, and first of all i ask apologize to the creator of this, that i am representing this as an issue, but i am really trying to get some help with this code and library, to modify or rewrite as a northern light show, so please for any idea, advice or hint would be very much appreciated.
Thank you in advance.

AlexMeadows, 2017-09-08:
How much of a start have you gotten?

I would look in colorschemes.py and make a new class that mimics one of the classes in there. The Rainbow color cycle (and all the others) would be good to look at since it has some of the fading of colors you will need. A function will be needed to fades between colors that you specify. How complicated it is will depend on how random you want the cycles of the light show to be and a lot of other things; however, you can start with smaller pieces and make it more dynamic and cool over time. Some things to think about are: how you define the ranges for the pulses of light(will they shift around, change sizes), how you determine and generate the colors, how you make new pulses start and change positions.

I have some ideas of how to do what you want, but I will wait to type them out unless you need the help.

tinue, 2017-09-09:
@raspberrypilights No problem; There isn't a "discussion" section on Github, so opening an issue is fine. Do you have a video of what you have in mind?

"Northern light" sounds more like a big flat x/y surface to me; So far the library is for a linear strip of lights, so you might need some adaptations in this respect as well.

raspberrypilights, 2017-09-11:
@AlexMeadows  thank you for your replying, yes I was looking and playing around with this colorschemes.py and rainbow function, than I created a brightness function so it would create an effect like fading away and than bright again, just like aurora but unfortunately my python skills are not so great and the second problem is about documentation of this lights, that's why I was trying to modify Mr. @tinue code rather than writing a new one from scratch. I have some videos but I need to figure it out how to upload the video.
@tinue  what I have in mind is selecting range of colors from green to purple (not strong purple) and add two more variables, speed and brightness, and start the spectre of  colors from greet to purple in the same time start speed and brightness in a random function, because also in the reality northern lights does not have any pattern, just random in the sky.
Sorry if I brought some confusion, I tried to be as clear as I can and lets hope that will be an interesting project. Thank you @AlexMeadows  and @tinue .

AlexMeadows, 2017-09-15:
@raspberrypilights [Here is a stack overflow question](https://stackoverflow.com/questions/41793471/python-rgb-led-color-fading) where someone is trying to make a function that fades between two colors. It seems like what you are wanting. You should be able to use the logic from it to modify the color wheel function to return the rgb value based on the current step of the pattern.

I found an [interesting website](http://www.perbang.dk/rgbgradient/) that might be helpful to visualize what the modifications to the color wheel need to do. It lets you define start and end rgb codes and it shows the gradient and rgb values.

I might work on a function that fades between two colors if I can find the motivation this weekend...it could be a nice addition to my light system.

Below is a snippet of code for a fade color cycle I made. It should be helpful to the fading the brightness in and out. I might update my public fork with changes I have made if I feel like it this weekend.
```
class Fade(ColorCycleTemplate):
    """Fades a color in and out."""
    def init(self, strip, num_led):
        self.direction = "up"
        self.brightness = 0

    def update(self, strip, num_led, num_steps_per_cycle, current_step,
               current_cycle):
        for pixel in range(num_led):
           strip.set_pixel_rgb_str(pixel, self.color, self.brightness)
        if (self.direction == "up"):
           self.brightness+=1
           if (self.brightness >= 100):
              self.direction = "down"
        else:
           self.brightness-=1
           if (self.brightness <= 0):
              self.direction = "up"
        return 1
```

raspberrypilights, 2017-09-18:
@AlexMeadows. Thank you buddy for the help, one problem less now. Propably i need to upload this project with the updates. And see my mess. Seems like we are facing the same problem with the motivation in weekends ;) anyway thanks

raspberrypilights, 2017-10-03:
@AlexMeadows i did tested your code for fading and works good but it is not smooth, seems like it have a second gap between action, brightness++ or --, do you think is any solution to make it pass smooth?

AlexMeadows, 2017-10-03:
Doing something like the code snippet below will make it fade smoother. Fading off the brightness bit is not as smooth as fading off of the rgb value. Also brightness does not scale evenly with the rgb values so you will want to speed up the fading when over roughly 40% brightness.

`   def set_pixel_rgb_str(self, led_num, rgb_color, bright_percent=100):
        """Sets the color of one pixel in the LED stripe.

        The changed pixel is not shown yet on the Stripe, it is only
        written to the pixel buffer.
        Colors are passed combined (3 bytes concatenated)
        If brightness is not set the global brightness setting is used.
        """
        lv = len(rgb_color)
        rgb = tuple(int(rgb_color[i:i + lv // 3], 16) for i in range(0, lv, lv // 3))
        self.set_pixel(led_num,
                       int(rgb[0]/100*bright_percent),
                       int(rgb[1]/100*bright_percent),
                       int(rgb[2]/100*bright_percent))`

closed 2018-03-09 by tinue