tinue/apa102-pi #25: Constant lights

bemn67, 2017-09-15:
This isn't really an issue but I could not find a way to message you! Sorry!

I was wondering if there is a way to have my LED strip constantly illuminated or for example have a constant rainbow cycle going. I have created a webserver to control the strip but now I need to figure out the Dotstar side of the programming and I am a little confused. I just need it to have a constant color across the whole strip and a cycle infinitely looping.

Any help is greatly appreciated, thanks!!!

AlexMeadows, 2017-09-15:
I made a few new files to add threading and stuff so I can control strips from a webserver on the same network. You should look at [this](https://github.com/AlexMeadows/APA102_Pi/tree/lightServer).

I have it listening on a port for json messages that tell it to do whatever pattern for whatever length of time or number of iterations. I have a ruby web server that send the messages to it. I can send you the web stuff I did if you want. I have been too busy to get around to cleaning it up enough make a repo and post publicly.

bemn67, 2017-09-15:
I have the whole webserver backend and frontend done, now I was just wondering how to keep a cycle infinitely looping. I have looked at your code and it looks like you just use the integer, -1, to continuously loop cycles, is that right? 

Thanks for the quick response!

AlexMeadows, 2017-09-15:
Yes, by default you can pass -1 for num_cycles and it will loop infinitely. With my code I did the same thing and made it so I can specify the time to stop at rather than cycles to run for and -1 for the time also makes it run infinitely.

bemn67, 2017-09-16:
Okay thank you so much! One more question I think, how could I stop a infinitely running cycle through code (not with ctrl+c)? For example I would have a second script running that would check my webserver to see if the strip color/mode has changed and if it has, stop the running cycle and update it with the new color and mode.

Thanks for all the help!

AlexMeadows, 2017-09-16:
The answer I had to that was with the threading and stuff I did in [lightServer.py](https://github.com/AlexMeadows/APA102_Pi/blob/lightServer/lightServer.py), [lightServerBackend.py](https://github.com/AlexMeadows/APA102_Pi/blob/lightServer/lightServerBackend.py) and a handful of minor changes in [colorcycletemplate.py](https://github.com/AlexMeadows/APA102_Pi/blob/lightServer/colorcycletemplate.py).

I made it so the APA102 stuff runs in one thread and I have a thread safe method of interacting with it from the main thread in which have a port listening for json commands. I used a port since it was super easy to send commands to the multiple PIs I have with lights, but you can change that part of lightServer.py to use whatever method you like to communicate with your web server.

lightServerBackend.py let me have one class I initialize that will start any color cycle depending on what I pass in, so lightServer.py stays more streamlined.

erebozz, 2017-10-03:
Hello
I am very confused, How can I change the colour of the Led Strip?

jmb, 2017-10-08:
@erebozz Once you have setup the strip, you can just use the `set_pixel(led_num, red, green, blue, bright_percent=100)` and `show` methods on it.

Something along the lines of:

```
import apa102
numLEDs = 10

strip = apa102.APA102(num_led=numLEDs, global_brightness=31, order='rgb')

for pixel in range(numLEDs):
    strip.set_pixel(pixel, 255, 0, 0) # Sets each pixel to red

strip.show()
```

closed 2018-03-09 by tinue