tinue/apa102-pi #27: Raspberry pi 3 apa102 ambilight

marc8219, 2017-11-08:
[image]

Will this work?
I do know the grapper, av converter and splitter they are right. And connect led strip to pi.

Extra parts:
12v to 5v 3a: https://www.ebay.de/itm/263062629260 (Extra question does it have to come from at power supply that has 12v 3a? Or is it ok with 12v 1.2a?)
Usb Hub: https://www.ebay.de/itm/302471693929
Hdmi Splitter: https://www.amazon.de/gp/product/B019MRGGJU/ref=ox_sc_act_title_1?smid=A2YXN6MAL2SF51&psc=1
Av Converter: https://www.amazon.de/gp/product/B01EIY768C/ref=ox_sc_act_title_3?smid=AGJ2TI2R2YFK8&psc=1
Grabber: https://www.amazon.de/gp/product/B0013BXFLG/ref=ox_sc_act_title_4?smid=A3JWKAKR8XB7XF&psc=1
Strips APA102: https://www.ebay.de/itm/222180982345?var=521020771678 (1m x2)

Any recommendations please tell, and im running on budget.
(Sorry if English is bad at some points)

AlexMeadows, 2017-11-08:
- The APA102 draw max of about 10 W per meter with 30 LEDs/Meter. 20 Watts at 5 Volts come out to 4 amps. I would get a larger voltage converter thing since it would not be able to power the lights at full brightness. Having extra capacity is good since power converter things tend to get hot near their limits.
- For the 12v and amps stuff, there is math with how efficient the converter to take into account for what it can output as a result of the input volts and amperage. 12v at 1.2a should do about about 5v 2.4a. 
- You will need a 3.3v to 5v Logic Level Sifter since the pi outputs at 3.3 volts and the strip will be powered by 5 volts. The voltage difference will make it not work properly because electronics and logic levels with voltage comparisons. Example of what you need: [https://www.adafruit.com/product/735](url)
- I would power the logic level converter off the same power supply as the led strip.

I don't really see what you are trying to do with the video converter and stuff, but the above is what I noticed from the led strip related things.

tinue, 2017-11-09:
I wonder why you use a 12V supply and then a converter? I did use such a converter in an older project (https://youtu.be/FNbzufNIeCg), but there my power source was a LiPo battery, so I had no choice.

I would recommend 1-2 pieces of a 5V 5A power supply instead.

About the video stuff I am also at a loss, I never tried something like an Ambilight.

closed 2017-11-09 by tinue

AlexMeadows, 2017-11-09:
I also would normally get a 5v power supply. Sometimes if I don't care about size I use an old computer power supply.