tinue/apa102-pi #30: Software SPI (Bit Bang) informations

kartilana, 2018-03-24:
Hi,
Can you please provide more informations in your [setup section](https://github.com/tinue/APA102_Pi/blame/master/README.md#L57) about how to activate Software SPI mode?
I've also seen in the [driver module](https://github.com/tinue/APA102_Pi/blob/master/apa102.py#L74) that mosi and sclk are hard codded. Can you explain how to change it in order to use the Software SPI instead of the Hardware SPI?
Thanks a lot for the job you've made, and thank you for your reply.
Regards.

tinue, 2018-03-25:
Hi, thanks for your interest in the library!
Since #3 I have changed the library to also support software SPI, so this is no longer an issue.

If you only have one single LED, the I suggest you start with sample.py. All you have to do is use the correct pins for MOSI and SCLK. If the colors are wrong, then you might also have two change the color ordering from 'rbg' to (often) 'rgb'.

The values in the driver are merely the default values in case nothing is passed when initializing the driver. There is a hardcoded assumption in the driver though: 10/11 is hardware SPI, anything else is software SPI. This is why the driver officially only supports the Raspberry Pi. The assumption is correct on this device.

kartilana, 2018-03-26:
Thanks a lot for such a quick reply!
I'll try to follow your advice with sample.py.
By the way, i forgot to mention that i try to make it work with a NanoPi NEO. So let's hope that will make it...

closed 2018-03-26 by kartilana