tinue/apa102-pi #36: Running sample with a single LED

irtemed88, 2019-08-21:
Firstly, thank you so much for building this library!

I have an APA102 LED that I soldered to a breakout board and wired to a breadboard following the diagram provided in your repo. I'd like to be able to set the color and brightness of just this one LED, but haven't been able to so far. 

I thought maybe there was an issue with my soldering, but I soldered and tested 3 different LEDs. For each one, if I execute `python3 runcolorcycle.py` I can see the single led lighting and changing _very_ dimly so I don't think it's an issue with my soldering.

Ideally I'd like to be able to execute `python3 sample.py` and the LED turn on for the duration of `time.sleep()`, but that has also been ineffective.

Here's how I modified the code of `sample.py`

```
#!/usr/bin/env python3
"""Ultra simple sample on how to use the library"""
from driver import apa102
import time

# Initialize the library and the strip
strip = apa102.APA102(num_led=1, global_brightness=10, mosi=10, sclk=11, order="bgr")

# Turn off all pixels (sometimes a few light up when the strip gets power)
strip.clear_strip()

# Prepare a few individual pixels
strip.set_pixel_rgb(0, 0xFF0000)

# Copy the buffer to the Strip (i.e. show the prepared pixels)
strip.show()

# Wait a few Seconds, to check the result
time.sleep(5)

# Clear the strip and shut down
strip.clear_strip()
strip.cleanup()
```

Am I making any sort of mistake here? I can provide a wiring diagram for my LED (which includes a level shifter for the data and clock lines).

Thanks again!

tinue, 2019-08-21:
The code is fine, it does light one LED on my strip. Since this is just one LED, I would start by wiring it directly to the Pi. Get 5V from the Pi GPIO header, and directly connect clock and data to the 3.3V pins of the Pi: It often does work without a level shifter.

I hope you did enable SPI using raspi-config.

Let me know if this works!

irtemed88, 2019-08-22:
Hi @tinue - Thanks for the quick reply and for validating the code isn't the issue. I have SPI enabled.

I took the level shifter out of the circuit, swapped to a different LED, and it _almost_ works. 

The first time I run my code the light now will turn on. The light is not the correct color and when `sample.py` completes the light remains on. Subsequent executions of `sample.py` have no effect unless I manually unplug both VCC and GND. After resetting VCC and GND manually I can run the script again and the light will turn on (albeit still with the wrong color). I changed the "order" property to "bgr" since that's what the docs for my LED say: http://www.led-color.com/upload/201604/APA102-2020%20SMD%20LED.pdf

I have a feeling maybe my solder points aren't well done which is causing strange behavior. I'm going to try a few more LEDs to ensure that isn't the issue.

irtemed88, 2019-08-22:
Turns out the issue was the soldering. Thank you for your time helping me debug! I guess this is why I mostly stick with software 😅

tinue, 2019-09-20:
Thanks for your feedback!

closed 2019-09-20 by tinue