tinue/apa102-pi #39: package installation location

xxv, 2019-11-14:
When installed via pip:

```
pip install apa102-pi
```

the installed libraries end up in the top-level packages `driver` and `colorschemes` instead of `apa102`:

```
$ ls lib/python3.7/site-packages/driver 
apa102.py  colorcycletemplate.py  __init__.py  __pycache__
```

These should probably be in an `apa102` package to avoid namespace conflicts.

tinue, 2019-11-16:
You are right, thanks! I'll fix this as soon as I can.

tinue, 2019-11-23:
If you have time, would you mind checking the new package? It can be installed from the test PyPi repo: `sudo pip3 install -i https://test.pypi.org/simple/ apa102-pi==2.3.0rc1`.
For safety, you might want to uninstall the previous version first: `sudo pip3 uninstall apa102-pi`

xxv, 2019-11-24:
Looks like it put it in the right spot! :+1:

tinue, 2019-11-24:
Thanks! I released this version now.

closed 2019-11-24 by tinue