tinue/apa102-pi #44: global_brightness range

stickperson, 2020-09-04:
Hi. Thanks for putting together such an awesome library! What is the acceptable range for `global_brightness`? Is it 0-255?

thetrancefan, 2020-09-10:
I'm not the author but I work with APA102 LEDs all the time. The range for brightness on those LEDs is 0-31, where 0 will turn all the LEDs off. The brightness (a five-bit integer) is embedded in the four bytes that are sent to each LED when you do a showI().

tinue, 2020-09-11:
Sorry, for the late reply, and thanks @thetrancefan for your answer. Both the question and the answer made me realise that the global brightness does not work as I intended.

On every individual pixel one can set both a color (RGB value), and a brightness for this pixel. The pixel brightness is a 5 bit value, so it goes from 0 to 31. The driver takes whatever value has been set and caps off all the bits except for the last 5. So 31 or 255 would both result in "full" brightness of 31.

There is no global brightness on the hardware, because all pixels work on their own. The global brightness is a thing of the driver. Its intent was to lower the strain on the power supply, by setting it to a conservative value. The comment in the code makes this clear.

By using a default of 31, I inadvertently already used the full value, though. I assumed that the value goes from 0 to 255...

Also, there is an additional brightness variable bright_percent if the set_pixel method.
I will have to rethink this entire brightness business and document it better. I am keeping this issue open until this is finished.

tinue, 2020-12-31:
Fixed with the latest release

closed 2020-12-31 by tinue