tinue/apa102-pi #45: User hardware SPI

edgeball, 2020-12-03:
I just can't get it to work

My Raspi 3 B v1.2 says 

> use hardware SPI

when I try to run the .py script

I did enable SPI in the config

KAN-PC046, 2020-12-04:
did you wire the LED strip directly to the pi or did you use a level shifter of some kind ?  
  
if you wired it directly its possible the leds cant read the signal from the pi  
Because the APA102 LEDs are 5 Volt and the Raspberry Pi is 3.3 volt  
  
if you are using a level shifter what kind are you using and how did you wire it ?  

some more info [here](https://github.com/tinue/apa102-pi#wiring) ?
```
The Raspberry Pi is a 3.3 volt device, and the APA102 LEDs are 5 volt devices. Therefore, it is possible that the first LED driver chip does not properly detect the 3.3 volt SPI signal from the Raspberry Pi.
The first chip will amplify the signal to 5 volts before passing it on, so it is really only the first chip that must detect the 3.3 volt signal.
To avoid this risk, use a 74AHCT125 or 74AHC125 level shifter for both the clock and the MOSI signal. In my limited testing with four different stripes from various Chinese sources I had no issues without a level shifter, but your experience might be different.
```

tinue, 2020-12-04:
The output is as it should be. The "Use hardware SPI" ist just a sort of info/debug output and not an error message. It looks like you have all of the software installed correctly. As @KAN-PC046 said, you are most likely looking at a hardware issue. This could really be anything, unfortunately. Broken LEDs, wrong type of LEDs, wrong wiring, bad soldering, level shifter required, ...
The most basic thing that often goes wrong is a mixup of the two data signals. Have you tried to swap MOSI and SCLK?

ulrischa, 2020-12-29:
Same problem here. I have checked all the connections and use an extra power supply for the dotstar strip. I use a rapberry zero WH. Could that be a problem?

KAN-PC046, 2020-12-29:
@ulrischa using a  rapberry zero WH can not not be a problem.

i think its 1 of 2 problems the GPIO pin is broken, The GPIO pins are 3.3V only and putting 5V on there wil probably break that pin
you can test this by using a multi meter or if you dont have one a [simple LED](https://thepihut.com/blogs/raspberry-pi-tutorials/27968772-turning-on-an-led-with-your-raspberry-pis-gpio-pins)

the 2nd option: the signal is not received properly to solve this use a 3.3V to 5V level shifter like the [74AHCT125](https://www.adafruit.com/product/1787)

ulrischa, 2020-12-29:
Ok it worked. I used this tutorial https://pimylifeup.com/raspberry-pi-led-strip-apa102/. The wires are not connected correct here.

plisker, 2020-12-30:
@ulrischa Hey, I'm working on this today, and using that tutorial myself. I don't exactly follow - what's was the issue with the tutorial / which wire is not connected correctly?

ulrischa, 2020-12-30:
@plisker the yellow and green cables habe to be changed
[image]

tinue, 2020-12-31:
@ulrischa Thanks for asking and supplying the picture. This seems to be a common problem. I added a troubleshooting tip in the readme that will hopefully help in the future. Right now it is only visible on the "develop" branch.
@KAN-PC046 Thanks for supporting. I often test with a Raspberry Pi Zero W and can absolutely confirm that it works fine.

closed 2020-12-31 by tinue