tinue/apa102-pi #47: Use hardware SPI

ranjitkathiriya, 2021-02-05:
Hey everyone,

I am using Raspberry pi 3 model B and below I am showing my output while running program.
```
pi@raspberrypi:~ $ python3 ./runcolorcycle.py
Three Seconds of white light
Use hardware SPI
Go twice around the clock
Use hardware SPI
One strandtest of red, green and blue each
Use hardware SPI
One slow trip through the rainbow
Use hardware SPI
Five quick trips through the rainbow
Use hardware SPI
Finished the test
```

I am getting this output but in my led nothing is happening. I have tried changing the pins green and yellow I am not getting any results. Can you please help me where I am wrong?

Thanks in advance.

tinue, 2021-02-06:
Hello, please have a look at the develop branch README file [(README.md)](https://github.com/tinue/apa102-pi/blob/develop/README.md). It contains additional troubleshooting recommendations for your case.
Regards, Martin

ranjitkathiriya, 2021-02-10:
Thanks! I have checked readme file after removing the Ground wire from raspberry its work fine. I don't know why it is so?

Actually, runcolor.py file and sample file is working properly, but I am not able to turn on a particular pixel-like LED1 = RED, LED2=BLUE.

tinue, 2021-02-10:
That's interesting... I am not an electrical engineer, so I can't explain what is going on here. 
For your other problem: Not sure if I understand. This is more or less what sample.py is doing (LED 12 RED, LED 24 WHITE and LED 40 GREEN). If you say that this one works, then what does not work?

ranjitkathiriya, 2021-02-11:
Thanks for the reply!

Sorry! It was my misunderstanding. Actually, just runcolor.py code is working, but it is blinking properly. About the sample.py when I run this code, all lights stay in white colour, and some LED change colours randomly each time I run the code.

According to your experience, it may be an LED issue or Raspberry pi issue? I think code is perfect.

closed 2021-02-11 by ranjitkathiriya