tinue/apa102-pi #52: Bit Banging Not Working

Lee-Stratford, 2022-05-30:
Hi

I'm having problems driving APA102 LED using the second SPI port (GPIO20 (pin 38) & GPIO21 (pin 40)) on a Raspberry Pi 4 v1.4 using Bullseye OS 64 bit. Second SPI port activated in config.txt as dtoverlay=spi1-1cs. 

I used to be able to use the second SPI port on previous versions of Raspian (not used APA102 in quite a while now), but now I can't. 

I changed the mosi and sclk to use the second SPI port, but it ignores the bit banging and uses the first SPI port (GPIO10 & GPIO11) instead. 

strip = apa102.APA102(num_led=5, mosi=20, sclk=21, order='rgb') # Initialize the library and the strip
strip.set_global_brightness(2)
strip.clear_strip()

Does anyone else have this problem?

Lee

tinue, 2022-05-31:
Is it bitbang or hardware SPI that you want to use? Also, I think you use the API the"old" way, and this no longer works.
For SPI bus 1 (2nd bus), you would use:
strip = apa102.APA102(num_led=5, spi_bus=1)

I can't test at the moment, but I think with the above initialisation it should work.

Lee-Stratford, 2022-05-31:
Hi

 

Thank you for your reply.

 

I am using Hardware SPI port 1 mosi & sclk connected to pins 38 & 40 of the GPIO on the RPi4. I am using a SSD1322 OLED on SPI port 0. 

 

Config.txt contains

 

# enable second SPI port

dtoverlay=spi1-1cs

 

Listing available SPI ports

ls /dev/spi*

/dev/spidev0.0  /dev/spidev0.1  /dev/spidev1.0

 

I have tried 

 

strip = apa102.APA102(num_led=5, spi_bus=1)

 

But Python spits out an error 

 

  strip = apa102.APA102(num_led=10,spi_bus=1) # Initialize the library and the strip

  File "/usr/local/lib/python3.9/dist-packages/apa102_pi/driver/apa102.py", line 104, in __init__

  self.check_input(bus_method, global_brightness, mosi, num_led, order, sclk, spi_bus, spi_ports)

  File "/usr/local/lib/python3.9/dist-packages/apa102_pi/driver/apa102.py", line 163, in check_input

  raise ValueError("Illegal spi_bus not in %s" % list(spi_ports))

ValueError: Illegal spi_bus not in [0, 6, 2, 3, 4, 5]

 

I have uninstalled the library using ‘sudo pip3 uninstall apa102-pi’ and reinstalled using ‘sudo pip3 install apa102-pi’, but still get the same error above. If I use mosi=20, sclk=21 instead of ‘spi_bus=1’ it should work?

 

Best Regards

 

Lee

tinue, 2022-05-31:
Can't say I understand it, but when I check the source code of the Adafruit library that lies underneath of apa102-pi, it appears that you need to use spi_bus number 6 for pins 38/40. Please give this a try.
It's most likely a typo in their library...

Lee-Stratford, 2022-05-31:
I tried it and got the same result. Also tried everything between 0 – 6 but just get 

 

OS error: /dev/spidev2.0 does not exist etc

 

Regards

 

Lee

tinue, 2022-05-31:
I fixed the typo in the Adafruit library, and now it's working on SPI bus 1. If I get to it I'll raise a pull request with them to get it fixed.
In the meantime, the file to edit is:
`.venv/lib/python3.9/site-packages/adafruit_blinka/microcontroller/bcm2711/pin.py`

That is if you installed as recommended into a virtual env. Otherwise you will have to find the file by yourself. 

At the end of the file you will see the spiPorts definition. Simply change the 6 to a 1 and save.

Lee-Stratford, 2022-05-31:
Hi

 

Great, I will have a play as I’m not really a Python programmer, just a basic noob lol but know my way around. 

 

I’m surprised that this issue hasn’t been picked up beforehand. I’d be grateful if you could raise the issue with Adafruit and get them to sort it out.

 

Many Thanks for helping 😊

 

Lee

Lee-Stratford, 2022-10-11:
Bingo! got it working on SPI1 after changing pin.py 😊

 

Just one thing, I keep getting ‘Use hardware SPI’ displayed in the Python Shell. Is there anyway of disabling this notification?

 

Best Regards

 

Lee

tinue, 2022-10-14:
Good to hear. Adafruit accepted by pull request a long time ago. Possibly if you get their latest version of the library, the issue is gone without patching.
About the "hardwares pi": Please check the "debug" branch. The output will not be there.

tinue, 2023-01-14:
This should no longer be an issue, because the Adafruit libraries have been updated.

closed 2023-01-14 by tinue