tinue/apa102-pi #58: LED remaining dark

Quaso17, 2023-10-24:
Hello everyone,

I am currently unable to control my LEDs even though it should work just fine as I double checked the wiring, energy supply and code. 
While debugging Ive encountered something strange. I was trying to run "sample.py" (only changed led count so it fits with my strip)  and while debugging I realized that when inside "strip.show()" there should be data that is send but there simply is none. I theorized that the problem could lie within the MOSI or SCLK Pins (10 and 11) but that doesnt seem to be the problem either unless im missing something. When going to line 355 in apa102.py to the "send_to_spi" method it is said to output data to the chosen SPI device. Im not sure if it is supposed to skip everything and end up in "else" but I feel like there is something wrong here. If anyone has any ideas please feel free to share them thank you.

Quaso17, 2023-10-24:
Another quick note: My chip enable value is none and I have no clue how to change that.

tinue, 2023-10-24:
Hello, first thing to check on a fresh Raspberry Pi installation: Is SPI enabled?
> If hardware SPI is used: SPI enabled and active (raspi-config, Interface Options, SPI, <Yes>); The SPI must be free and unused.
Regards, Martin

Quaso17, 2023-10-24:
SPI is enabled, yes. When I was using an Arduino Uno things seemed to work just fine.

tinue, 2023-10-24:
OK, in show(), the variable self.leds contains the pixel data. If should be an array with the size of the led count.
send_to_spi seems ok, the else case uses hardware spi which will be used if you don't select anything else. The chip enable can normally be ignored, because these strips have no pin for it.
It might be possible that you have a strip that can't deal with 3.3 volts for the signaling voltage, and you need to add a voltage converter. I have never seen a strip actually needing this, but it certainly possible. An Arduino Uno has 5V on its pins.

Quaso17, 2023-10-24:
I do infact use an external power supply for the 5V. The last time I didnt the Raspberry couldnt handle it. The LEDs arent broken either so I am still struggling to think of a what the issue could be. My only guess would be that the MOSI and SCLK Pins arent working properly but I dont know why they wouldnt it also seems odd to me.

tinue, 2023-10-24:
SCLK and MOSI still use 3.3V on the Raspberry Pi, and the Strip expects 5V. Technically the strip triggers a "high" somewhere between ground and the power supply voltage. The data sheets typically say 70%. So if your power supply delivers 5V, 70% would be 3.5V. The 3.3V of the Pi would thus still be a "low". As I said, despite the data sheets it seems to work with 3.3V on most strips.
You can try to lower the voltage of your power supply. If it is on 4.7V, then 70% are below 3.3V, and thus 3.3V should register as "high".

Quaso17, 2023-10-24:
Unfortunately lowering it to 4.7V had no effect. Is there a way to check if my MOSI and SCLK Pins are intact?

tinue, 2023-10-24:
Apart from using some other device that is controlled via SPI, I don't think there is an option. I hope you tried to swap MOSI and SCLK, because I always get them wrong at the first attempt.
Do you power the Pi from the same power supply? If not, then you must make sure that the Pi and the Strip share a common ground (e.g. connect the Strip / power supply ground with one of the ground pins of the Pi).

Quaso17, 2023-10-24:
I have swapped them yes, but im pretty sure I got it correct anyway. And I use a different power supply for the Raspberry but made sure to use the same Ground. All things considered there is no reason for the LEDs to not work. Ive been trying to figure out the wrongs for 2 weeks now. Could the Raspbery Version or a specific OS be the root of the problem? Thats my only guess left but I dont think that idea holds any water either.

tinue, 2023-10-24:
PI or OS should not have an impact. Whenever a new release came out I performed a quick test, including the latest one (October 2023). I do not have a Model 5 yet, but I tested pretty much every older one as well.
Do you have another strip, or maybe even just a single APA102 LED? They are on many devices and the only problem is usually to find out on which PIN they expect MOSI and SCLK.

One more idea: Use some other GPIO pins for MOSI and SCLK, and then use Software Bit Bang mode. For example runcolorcycle_tree.py uses pins 12 and 25.

Quaso17, 2023-10-24:
It actually does work this way, thank you very much. But how come it works only in this constellation?

tinue, 2023-10-24:
It would indeed point to a broken SPI on your Pi. You could also try another SPI Bus: The Raspberry Pi 4 hat four of them, as far as I know. You would have to figure out how to enable them and which pins they are on, though, as I have not done this so far.

Quaso17, 2023-10-24:
It seems so, thank you agan for your help. There are just two more question I have. I have a strip of 36 LEDs but when I change NUM_LED to said number the strip goes wild and I dont know why. The other one would be that I dont see a way to configure the colors for runcolorcycle_tree.py. Cant I use set_pixel_rgb there? Or can I?

tinue, 2023-10-24:
There might still be a voltage issue. I looks like the strip does not receive proper and strong data. I would recommend to use a level shifter as outlined in the readme. The runcolorcycle scripts are just samples that run a number of patterns, there is nothing configurable there. 
To do your very own thing, start with sample.py.

Quaso17, 2023-10-26:
The wires were a bit loose so I just needed to fix that. Everything is running smoothly now.

closed 2023-10-26 by Quaso17