tinue/apa102-pi #60: lgpio.error: 'GPIO busy' in test script, R.Pi 5

breaker-of-stone, 2024-08-19:
Previous work on an R.Pi.4 is throwing an error on the R.Pi.5.

I'm using apa102_pi and the demo script therein:
https://github.com/tinue/apa102-pi/blob/main/runcolorcycle.py

and seeing a 'GPIO busy' exception. Configuration and exception dump follows. How to diagnose?

pinout
```
Description        : Raspberry Pi 5B rev 1.0
Revision           : d04170
SoC                : BCM2712
RAM                : 8GB
```

lsb_release -a
```
No LSB modules are available.
Distributor ID:	Debian  # raspi-OS via R.Pi Imager
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

python --version
`Python 3.11.2 # distro
`
venv is engaged

Prior to this situation I encountered:

```
  File "/noisebin/.venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```

So chose to:

```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpio
```

/boot/firmware/config.txt has:
```
dtparam=i2c_arm=on
dtparam=spi=on

[all]
dtoverlay=spi1-3cs
```

Have enabled SPI in raspi-config for good measure.

pinctrl gives:
```
116: ip    pd | lo // GPIO16 = input
117: ip    pd | lo // GPIO17 = input
118: ip    pd | lo // GPIO18 = input
```

gpioinfo says:
```
line  16:     "GPIO16"   "spi1 CS2"  output   active-low [used]
line  17:     "GPIO17"   "spi1 CS1"  output   active-low [used]
line  18:     "GPIO18"   "spi1 CS0"  output   active-low [used]
```

Issue follows. Results are identical if running under sudo.

```
Traceback (most recent call last):
  File "/home/quiet-mode/examples/led-light-drivers/rcc.py", line 47, in <module>
    main()
  File "/home/quiet-mode/examples/led-light-drivers/rcc.py", line 14, in main
    my_cycle.start()
  File "/quiet-mode/.venv/lib/python3.11/site-packages/apa102_pi/driver/colorcycletemplate.py", line 71, in start
    strip = apa102.APA102(num_led=self.num_led, bus_method=self.bus_method, spi_bus=self.spi_bus,
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/quiet-mode/.venv/lib/python3.11/site-packages/apa102_pi/driver/apa102.py", line 130, in __init__
    ce = eval("digitalio.DigitalInOut(board.D" + str(ce) + ")")
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
  File "/quiet-mode/.venv/lib/python3.11/site-packages/digitalio.py", line 193, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/quiet-mode/.venv/lib/python3.11/site-packages/digitalio.py", line 223, in direction
    self._pin.init(mode=Pin.IN)
  File "/quiet-mode/.venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
  File "/quiet-mode/.venv/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 696, in setup
    _check(lgpio.gpio_claim_input(_chip, gpio, {
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/quiet-mode/.venv/lib/python3.11/site-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/quiet-mode/.venv/lib/python3.11/site-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

Counter-evidence: Another demo from the apa102_pi repo succeeds, has no errors:
https://github.com/tinue/apa102-pi/blob/main/sample.py

tinue, 2024-08-19:
Hello, thanks for the report. It's been a while since I last tried on a Raspberry Pi 5, but it worked at the time. I suspect a chance in Adafruit's Blinka library, which is imported by apa102. Unfortunately I can't test for at least a week, so a resolution might take a while.

breaker-of-stone, 2024-08-19:
Thanks, would at least welcome verification.

tinue, 2024-08-29:
Unfortunately I cannot reproduce the error. I have the same Pi 5 revision as you, and I have fully updated/upgraded the OS, and upgraded all Python packages. See the list below.

There is one difference in my setup: After installing apa102-pi, I uninstall it right away (I only want the required dependencies). Then, I use the cloned apa102-pi, and run e.g. `./runcolorcycle.py`. You seem to run e.g. `.venv/lib/python3.11/site-packages/apa102_pi/driver/colorcycletemplate.py`, while I would run `./apa102_pi/driver/colorcycletemplate.py`. Could you try this as well?

```
Package                                  Version
---------------------------------------- --------
Adafruit-Blinka                          8.47.0
adafruit-circuitpython-bitbangio         1.3.15
adafruit-circuitpython-busdevice         5.2.9
adafruit-circuitpython-connectionmanager 3.1.1
adafruit-circuitpython-requests          4.1.6
adafruit-circuitpython-typing            1.11.0
Adafruit-PlatformDetect                  3.73.0
Adafruit-PureIO                          1.1.11
binho-host-adapter                       0.1.6
certifi                                  2024.7.4
chardet                                  5.2.0
charset-normalizer                       3.3.2
colorzero                                2.0
distro                                   1.9.0
gpiozero                                 2.0.1
idna                                     3.8
lgpio                                    0.2.2.0
pigpio                                   1.78
pip                                      24.2
pycryptodomex                            3.20.0
pyftdi                                   0.55.4
pyserial                                 3.5
python-apt                               2.6.0
pyusb                                    1.2.1
requests                                 2.32.3
RPi.GPIO                                 0.7.1
rpi-ws281x                               5.0.0
setuptools                               74.0.0
six                                      1.16.0
smbus2                                   0.4.3
spidev                                   3.6
ssh-import-id                            5.11
sysv-ipc                                 1.1.0
toml                                     0.10.2
typing_extensions                        4.12.2
urllib3                                  2.2.2
wheel                                    0.44.0
```

breaker-of-stone, 2024-10-27:
Martin,

For closure I can confirm that I was successful after I followed the 'Full
Installation' procedure in the README document.

I am now trying to build a more flexible colour cycle component based on
the coloraide module:
        https://facelessuser.github.io/coloraide/interpolation/

I am building a public art project that children can interact with while
recycling drink containers.  It's complicated!
Thank you for your advice, it put me on the right track.
36 Celcius in Perth Western Australia this week.

breaker-of-stone, 2025-02-20:
Martin,
Ha!  My issue #60 (and your tactics to work around) appear to have been
triggered by a change in kernel policy for the RPi.5

Melissa at AdaFruit explains while resolving an issue with Blinka /
digitalio - long thread!:
https://github.com/adafruit/Adafruit_Blinka/issues/755

TL;DR: SPI bus Chip Select 0 on each bus is now marked as in use by the
kernel, on the RPi.5

Solutions include either removing the SPI dtoverlay and bitbanging it,
using an spi*-0cs overlay, or deceiving the kernel by assigning cs0 to an
unused pin but driving LEDs on cs1 and cs2.  I don't want to throw away any
pins though because our design needs every pin!

I am looking for an unused 'virtual' Broadcom line number that doesn't
connect to any external pin, to see if that also works.  Did this with
cs0_pin=45 on an RPi.4, where that BCM line wasn't used. Seemed to work, or
had no noticeable side effects.

Relentless in Western Australia 😉

tinue, 2025-02-24:
Thanks for the information, very interesting. Apparently my initial testing with the Pi 5 predated the change. When I tested again, Adafruit had already added the work around in their code. I never hit the issue, therefore.

closed 2025-02-24 by tinue

tinue, 2025-02-25:
No, I never paid attention to thread safety. I also don't know if the Adafruit libraries are thread-safe.
Given that strips do not have a CS input, I assume that you handle this with some hardware on your own, right?

breaker-of-stone, 2025-02-26:
Ok, short version:

The Adafruit libraries do not guard or lock the SPI bus at message level,
so two separate threads (or Linux processes) can send messages
simultaneously that become intermingled.  Whether they are thread-safe is
irrelevant in these circumstances.

This manifests as LED pattern data bits for one chip-select being sent to
LEDs that are activated by a different chip-select - i.e. visual crosstalk
and pattern corruption.

There was slight improvement in isolation when switching from the SPI bus
driver to a bitbang technique, but the effect remained.

One may have to write a low level SPI bus driver that is a gatekeeper for
one _logical_ channel at a time - something beyond my time capacity.

Our strategy may need to switch to stronger hardware separation.  The
Raspberry Pi main controller cannot spare any more pins for separate SPI
busses so we may delegate patterns to one smaller dedicated controller per
bus.  I have a spare TinyPico (ESP32 microcontroller) that looks capable.

I may post on the AdaFruit forums but not optimistic of a solution.

Regards, Ian